dafta/DeckMTP #2: USB-C to USB-C not working

Maxaker, 2023-02-28:
Hi,

just wanted to report that connection over a USB-C to USB-C cable doesn't seem to work.

dafta, 2023-03-01:
Are you sure? Is there a different cable you can try maybe?

Because I have done all my testing over a USB-C to USB-C cable and it works.

Maxaker, 2023-03-02:
I tried with that USB-A to USB-C 10 Gbps cable aswell, doesnt work either.

dafta, 2023-03-02:
Are you on Stable? Also, which BIOS version do you have?

Maxaker, 2023-03-02:
I am on stable and my BIOS is 4403 from here https://rog.asus.com/de/motherboards/rog-strix/rog-strix-x570-e-gaming-model/helpdesk_bios/

dafta, 2023-03-02:
Sorry, I was asking about the Deck's BIOS. You can see the version if you go to Settings > System.

Maxaker, 2023-03-02:
Ok nevermind that store thing, was because I had store plugin filter set to testing.

The problem is still happening, the switch pro controller USB cable works (I think its 2.0), all other cables that are 3.0 or above dont work, under devices the steamdeck shows up with "driver error".

dafta, 2023-03-02:
Do you have another PC you could try it on?

Maxaker, 2023-03-02:
I tested with a PC using the motherboard https://www.gigabyte.com/de/Motherboard/GA-Z97N-Gaming-5-rev-10#ov with BIOS Version F3 and it worked with the USB-A to USB-C 10Gbps cable. I only tested one USB 3.0 port on the back.

Only difference I can see for now is that my other pcs motherboard only has 10Gbps USB ports.

dafta, 2023-03-02:
Did you try the USB-C to USB-C cable on that other PC as well?

Maxaker, 2023-03-02:
I cant, it doesnt have a USB-C port.

Maxaker, 2023-03-02:
Just to clarify, on my regular pc I have problems both the 2 USB-C to USB-C cables that are USB4 and with the new USB-A to USB-C cable that is USB 3.1.

On that other PC the USB 3.1 cable worked.

dafta, 2023-03-02:
Okay. I'm not sure what I can do to help, seeing as how USB-C to USB-C works from everything I've tried. It might be an issue with your motherboard, or the OS installed on your PC, or it might be something else entirely. It's hard to tell what the issue might be, but given that the same cable works on one PC and doesn't work on the other, I'd be inclined to say that the fault is in the PC that doesn't work.

Maxaker, 2023-03-02:
I think it has to do with the combination of USB port and cable. Did you test any highspeed ports with a cable to match? like 10Gbps and above?

dafta, 2023-03-02:
Yeah, I have. I tested with several cables, but the fastest is probably the USB 3.2 cable that came with my Samsung T7 external SSD.

As for the ports, I tested it on my laptop, on all three of my ports. This includes two USB 3.2 gen 2 ports, and one USB 3.2 gen 1 port.

Maxaker, 2023-03-02:
I will try to test around a bit. Don't know if I will reinstall. I could try a live USB stick on that pc.

lbibass, 2023-03-06:
my NekTeck USB-C cable doesn't work with a 2021 MacBook Pro and the Deck.

mcnhm2, 2023-03-16:
The some issue here no matter what i do it won't work ;(

dafta, 2023-03-16:
> my NekTeck USB-C cable doesn't work with a 2021 MacBook Pro and the Deck.

Did you try with another cable maybe?

> The some issue here no matter what i do it won't work ;(

Does it work with any cable?

There are two separate issues here. Some combinations of SteamOS versions and BIOS versions just plainly don't work, so I'd urge you to test if other cables work or if it just doesn't work at all.

I'm currently unable to debug because I don't have my Deck for a while, since last month and up until maybe two weeks for now.

mcnhm2, 2023-03-17:
> Did you try with another cable maybe?
>
>
> Does it work with any cable?
>
> There are two separate issues here. Some combinations of SteamOS versions and BIOS versions just plainly don't work, so I'd urge [...]
I have try with many cable none of them work, it just say its charge slow on steam deck.

mcnhm2, 2023-03-17:
I have try with many cable none of them work, it just say its charge slow on steam deck.

dafta, 2023-03-17:
@mcnhm2 This is an unrelated issue, I'd urge you to post about this here: https://github.com/dafta/DeckMTP/issues/3

Also, if you'd post the dmesg as shown in my comment there.

TimestaticYT, 2023-03-19:
I've tried this on My M1 Mac pro and have the same issue

TimestaticYT, 2023-03-19:
Worked for a moment and now doesn't let me connect again

dafta, 2023-03-19:
@TimestaticYT Did you try different cables? If you have, could you please reboot the Deck, try again once, and send me the complete output of `journalctl -k -x`?

dafta, 2023-03-20:
When you do that, could you also include the version of SteamOS you're on (Stable, Beta, Main), and the BIOS version (you can find that under Settings > System)?

Honzackcz, 2023-03-27:
When I connect Deck to PC with USB-C to USB-C cable I can only copy files from Deck to PC. When I try to copy files from PC to Deck is it able to only create the file in Deck but cannot write to it and seem like the plugin crashes. USB-A 2.0 cable works fine both ways.

dafta, 2023-03-27:
@Honzackcz Would you be able to reboot, try this again with the USB-C to USB-C cable, and when it fails, post the complete output of the `journalctl -k -x` command?

Honzackcz, 2023-03-28:
Dunno if I did it right, anyway here is the log: 
[log.txt](https://github.com/dafta/DeckMTP/files/11085033/log.txt)

Honzackcz, 2023-03-28:
What I did:
- turned Deck on
- waited for complete boot to Steam game mode
- connected C to C cable
- enabled DeckMTP plugin
- tried to copy a file from WinPC to Deck - in middle of copying Deck disconnected from WinPC, the file is created with right name but contains 0 bytes
- went to desktop
- wrote the command to terminal and saved it's output to file

Also I noticed before that when the file transfer failed, the DeckMTP in plugin menu have still toggle in on state and when I clicked on it, it switched to off state but the plugin started to works again and connected to WinPC showing Deck's storage.

dafta, 2023-03-29:
@Honzackcz The log you posted is cropped to the size of your terminal. It doesn't show the full lines, and you can see that it says `lines 1-15` on the last line, meaning it's showing only the first 15 lines. Everything you did was correct, but I assume for the command that you just copy pasted everything that was visible at the time to the log file you sent.

This is my bad, I should have specified how to save the output to a file. Would you please repeat the same thing, using the following command?

`journalctl -x -k > log.txt`

After this, the whole log should be saved in your home directory, in the file `log.txt`.

Honzackcz, 2023-03-29:
ok, this should be all now
[log.txt](https://github.com/dafta/DeckMTP/files/11101098/log.txt)

dafta, 2023-03-29:
Did you try turning on and off the plugin? I don't see any logs from it.

Honzackcz, 2023-03-29:
I did it exactly same as before

Honzackcz, 2023-03-29:
Just an idea, could it be caused by the cable? The cable is rated for 5 Gb/s, Deck should be able to handle 10 Gb/s and PC 20 Gb/s, so if the USB chips don't recognize USB cable's speed and push 10 Gb/s thru it some data can be lost due to integrity issues.

dafta, 2023-03-29:
Ok, my bad once again. It should be `journalctl -x > log.txt`, without the `-k` argument.

As for the cable, it's possible but I doubt it. What's more likely is that it's a bug in the MTP responder, and the logs should show the problem.

Honzackcz, 2023-03-29:
[log.txt](https://github.com/dafta/DeckMTP/files/11102922/log.txt)

Mxm471, 2023-03-29:
I can transfer files with a 2.0 cable as Well. 3.2 does not Work. Crashes when i try to Copy Files onto the Deck unfortunately.

Honzackcz, 2023-03-29:
I haven't tried just USB 3.2 yet because I don't have A to C cable but it should arrive by the end of the week, so I can try it then too.

Honzackcz, 2023-03-31:
OK, I've got the USB 3.2 A to C cable and it does the same thing as C to C cable.
Here are logs after failed transfer from PC to Deck thru USB 3.2 A to C (logx is with -x parameter and logb is with -b):
[logx.txt](https://github.com/dafta/DeckMTP/files/11122469/logx.txt)
[logb.txt](https://github.com/dafta/DeckMTP/files/11122471/logb.txt)

dafta, 2024-06-18:
Potentially fixed by 1.0.3, needs testing.

dafta, 2024-06-18:
This is most likely the same issue as #9 , a problem with Windows and MTP drivers. I'm closing this issue to keep the discussion focused on one issue.

closed 2024-06-18 by dafta