dafta/DeckMTP #3: Possibly Faulty Deck

gameboycjp, 2023-03-11:
The plugin does not work on my deck, tested tow cables and two computers. DRD is already set. Sometimes while turning off mtp, the deck will reboot itself instantaneously. Nothing shows up in the system logs. Where would the plugin keep logs? Any other troubleshooting ideas?

JorgeEspadas, 2023-03-12:
Turning MTP off after disconnecting also insta-kills the deck if its lucky enough to reboot to a working state that is, i've had instances where it just locks it and i have to do a hard reset.

It is also mounting the internal SSD and SDCards as read-only wich makes this kinda useless.

dafta, 2023-03-14:
It's strange that it mounts the drives as read-only, that's not how it is supposed to work. As for resetting the Deck, I don't have an idea what might be causing it, and I'm currently not able to test, as I've been dealing with RMA issues for the past month. As soon as I get it back, I'll try and recreate the issue.

The plugin doesn't keep logs at the moment, but I will add that in a small update as soon as I get my Deck back, and then you can try it with that version and see what happens. However, there aren't really any useful logs the plugin might output.

What would be useful, is if you could send me the syslog from the crash. Try and induce the crash once again, and then when you boot the Deck again, send me the output of the following commands:

`journalctl -x -b -1`
`journalctl -k -x -b -1`
`dmesg`

cchrkk, 2023-03-17:
I have a similar problem, the plugin never woked for me (even previous version). DRD is enabled in bios. When I toggle mtp while connected to pc this happens but no device gets detected on pc:
```
Mar 17 15:45:08 steamdeck PluginLoader[4031]: mkdir: cannot create directory ‘cfg’: File exists
Mar 17 15:45:08 steamdeck PluginLoader[4032]: mount: /home/deck/homebrew/plugins/DeckMTP/bin/cfg: none already mounted on /home/deck/homebrew/plugins/DeckMTP/bin/cfg.
Mar 17 15:45:08 steamdeck PluginLoader[4032]:        dmesg(1) may have more information after failed mount system call.
Mar 17 15:45:08 steamdeck PluginLoader[4033]: mkdir: cannot create directory ‘cfg/usb_gadget/g1’: File exists
Mar 17 15:45:08 steamdeck PluginLoader[4034]: mkdir: cannot create directory ‘configs/c.1’: File exists
Mar 17 15:45:08 steamdeck PluginLoader[4035]: mkdir: cannot create directory ‘strings/0x409’: File exists
Mar 17 15:45:08 steamdeck PluginLoader[4036]: mkdir: cannot create directory ‘configs/c.1/strings/0x409’: File exists
Mar 17 15:45:08 steamdeck PluginLoader[4037]: mkdir: cannot create directory ‘functions/ffs.mtp’: File exists
Mar 17 15:45:08 steamdeck PluginLoader[4038]: ln: failed to create symbolic link 'configs/c.1/ffs.mtp': File exists
Mar 17 15:45:08 steamdeck PluginLoader[4039]: mkdir: cannot create directory ‘/dev/ffs-mtp’: File exists
Mar 17 15:45:08 steamdeck PluginLoader[4040]: mount: /dev/ffs-mtp: mtp already mounted on /dev/ffs-mtp.
Mar 17 15:45:08 steamdeck PluginLoader[4040]:        dmesg(1) may have more information after failed mount system call.
Mar 17 15:45:08 steamdeck PluginLoader[4041]: mkdir: cannot create directory ‘/etc/umtprd’: File exists
Mar 17 15:45:08 steamdeck kernel: ffs_data_put(): freeing
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] uMTP Responder
Mar 17 15:45:08 steamdeck PluginLoader[4046]: [uMTPrd - 15:45:08 - Info] uMTP Responder
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Version: v1.6.2 compiled the Mar 14 2023@16:49:51
Mar 17 15:45:08 steamdeck PluginLoader[4046]: [uMTPrd - 15:45:08 - Info] Version: v1.6.2 compiled the Mar 14 2023@16:49:51
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] (c) 2018 - 2021 Viveris Technologies
Mar 17 15:45:08 steamdeck PluginLoader[4046]: [uMTPrd - 15:45:08 - Info] (c) 2018 - 2021 Viveris Technologies
Mar 17 15:45:08 steamdeck PluginLoader[4046]: [uMTPrd - 15:45:08 - Info] Sending command : addstorage:/run/media/mmcblk0p1 "SD Card" rw
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Add storage SD Card - Root Path: /run/media/mmcblk0p1 - UID: -1 - GID: -1 - Flags: 0x00000000
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Add storage Home - Root Path: /home/deck - UID: 1000 - GID: 1000 - Flags: 0x00000000
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Device path : /dev/ffs-mtp/ep0
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB In End point path : /dev/ffs-mtp/ep1
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Out End point path : /dev/ffs-mtp/ep2
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Event End point path : /dev/ffs-mtp/ep3
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Max packet size : 0x200 bytes
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Max write buffer size : 0x2000 bytes
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Max read buffer size : 0x2000 bytes
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Read file buffer size : 0x100000 bytes
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Manufacturer string : Valve
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Product string : Steam Deck
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Serial string : FVAA15101FDB
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Firmware Version string : Rev A
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Interface string : MTP
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Vendor ID : 0x28DE
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Product ID : 0x3000
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB class ID : 0x06
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB subclass ID : 0x01
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Protocol ID : 0x01
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB Device version : 0x3008
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] USB FunctionFS Mode
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Wait for connection : 0
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Loop on disconnect : 0
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Show hidden files : 1
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] File creation umask : System default umask
Mar 17 15:45:08 steamdeck PluginLoader[4045]: [uMTPrd - 15:45:08 - Info] Default UID : 1000
[... cut ...]
Mar 17 15:45:08 steamdeck kernel: read strings
Mar 17 15:45:37 steamdeck PluginLoader[4063]: ./stop.sh: line 5: echo: write error: No such device
Mar 17 15:45:37 steamdeck kernel: unloading
Mar 17 15:45:37 steamdeck systemd[1]: home-deck-homebrew-plugins-DeckMTP-bin-cfg.mount: Deactivated successfully.
Mar 17 15:45:37 steamdeck kernel: ffs_data_put(): freeing
Mar 17 15:45:37 steamdeck systemd[1]: dev-ffs\x2dmtp.mount: Deactivated successfully.
```

I think I will investigate more with dmesg and report here. Should I open a new issue?

dafta, 2023-03-17:
Okay, so the first line about the cfg directory not existing is fine. The rest of the `file exists` messages shouldn't be there. Would you please reboot the Deck, then try again, and send the logs again?

cchrkk, 2023-03-17:
> Okay, so the first line about the cfg directory not existing is fine. The rest of the `file exists` messages shouldn't be there. Would you please reboot the Deck, then try again, and send the logs again?

Tried rebooting and the file exists wasn't there at the beginning but showed up after toggling mtp on and off some times / plugging unplugging usb cable/switching usb port on pc

```
Mar 17 21:59:54 steamdeck PluginLoader[3089]: mkdir: cannot create directory ‘cfg’: File exists
Mar 17 21:59:54 steamdeck PluginLoader[3090]: mount: /home/deck/homebrew/plugins/DeckMTP/bin/cfg: none already mounted on /home/deck/homebrew/plugins/DeckMTP/bin/cfg.
Mar 17 21:59:54 steamdeck PluginLoader[3090]:        dmesg(1) may have more information after failed mount system call.
Mar 17 21:59:54 steamdeck PluginLoader[3091]: mkdir: cannot create directory ‘cfg/usb_gadget/g1’: File exists
Mar 17 21:59:54 steamdeck PluginLoader[3092]: mkdir: cannot create directory ‘configs/c.1’: File exists
Mar 17 21:59:54 steamdeck PluginLoader[3093]: mkdir: cannot create directory ‘strings/0x409’: File exists
Mar 17 21:59:54 steamdeck PluginLoader[3094]: mkdir: cannot create directory ‘configs/c.1/strings/0x409’: File exists
Mar 17 21:59:54 steamdeck PluginLoader[3095]: mkdir: cannot create directory ‘functions/ffs.mtp’: File exists
Mar 17 21:59:54 steamdeck PluginLoader[3096]: ln: failed to create symbolic link 'configs/c.1/ffs.mtp': File exists
Mar 17 21:59:54 steamdeck PluginLoader[3097]: mkdir: cannot create directory ‘/dev/ffs-mtp’: File exists
Mar 17 21:59:54 steamdeck PluginLoader[3098]: mount: /dev/ffs-mtp: mtp already mounted on /dev/ffs-mtp.
Mar 17 21:59:54 steamdeck PluginLoader[3098]:        dmesg(1) may have more information after failed mount system call.
Mar 17 21:59:54 steamdeck kernel: ffs_data_put(): freeing
Mar 17 21:59:54 steamdeck PluginLoader[3099]: mkdir: cannot create directory ‘/etc/umtprd’: File exists
Mar 17 21:59:54 steamdeck PluginLoader[3104]: [uMTPrd - 21:59:54 - Info] uMTP Responder
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] uMTP Responder
Mar 17 21:59:54 steamdeck PluginLoader[3104]: [uMTPrd - 21:59:54 - Info] Version: v1.6.2 compiled the Mar 14 2023@16:49:51
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Version: v1.6.2 compiled the Mar 14 2023@16:49:51
Mar 17 21:59:54 steamdeck PluginLoader[3104]: [uMTPrd - 21:59:54 - Info] (c) 2018 - 2021 Viveris Technologies
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] (c) 2018 - 2021 Viveris Technologies
Mar 17 21:59:54 steamdeck PluginLoader[3104]: [uMTPrd - 21:59:54 - Info] Sending command : addstorage:/run/media/mmcblk0p1 "SD Card" rw
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Add storage SD Card - Root Path: /run/media/mmcblk0p1 - UID: -1 - GID: -1 - Flags: 0x00000000
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Add storage Home - Root Path: /home/deck - UID: 1000 - GID: 1000 - Flags: 0x00000000
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Device path : /dev/ffs-mtp/ep0
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB In End point path : /dev/ffs-mtp/ep1
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Out End point path : /dev/ffs-mtp/ep2
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Event End point path : /dev/ffs-mtp/ep3
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Max packet size : 0x200 bytes
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Max write buffer size : 0x2000 bytes
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Max read buffer size : 0x2000 bytes
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Read file buffer size : 0x100000 bytes
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Manufacturer string : Valve
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Product string : Steam Deck
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Serial string : FVAA15101FDB
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Firmware Version string : Rev A
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Interface string : MTP
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Vendor ID : 0x28DE
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Product ID : 0x3000
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB class ID : 0x06
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB subclass ID : 0x01
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Protocol ID : 0x01
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB Device version : 0x3008
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] USB FunctionFS Mode
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Wait for connection : 0
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Loop on disconnect : 0
Mar 17 21:59:54 steamdeck PluginLoader[3103]: [uMTPrd - 21:59:54 - Info] Show hidden files : 1
[... cut ...]
Mar 17 21:59:54 steamdeck kernel: read descriptors
Mar 17 21:59:54 steamdeck kernel: read strings
Mar 17 21:59:56 steamdeck PluginLoader[3111]: ./stop.sh: line 5: echo: write error: No such device
Mar 17 21:59:56 steamdeck kernel: unloading
Mar 17 21:59:56 steamdeck systemd[1]: home-deck-homebrew-plugins-DeckMTP-bin-cfg.mount: Deactivated successfully.
Mar 17 21:59:56 steamdeck kernel: ffs_data_put(): freeing
Mar 17 21:59:56 steamdeck systemd[1]: dev-ffs\x2dmtp.mount: Deactivated successfully.
```

dafta, 2023-03-18:
Okay, so this shows me the same information. If you toggled it on and off a couple of times, I assume this is from the last time. I need the logs from the first time you try toggling the plugin.

Could you please reboot the Deck again, toggle the plugin on, and send all the logs up to that point with `journalctl -k -x`?

cchrkk, 2023-03-18:
Ok, I've let the deck boot.
Toggled the switch:
```
Mar 18 09:04:25 steamdeck kernel: file system registered
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] uMTP Responder
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Version: v1.6.2 compiled the Mar 14 2023@16:49:51
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] (c) 2018 - 2021 Viveris Technologies
Mar 18 09:04:25 steamdeck PluginLoader[5123]: [uMTPrd - 09:04:25 - Info] uMTP Responder
Mar 18 09:04:25 steamdeck PluginLoader[5123]: [uMTPrd - 09:04:25 - Info] Version: v1.6.2 compiled the Mar 14 2023@16:49:51
Mar 18 09:04:25 steamdeck PluginLoader[5123]: [uMTPrd - 09:04:25 - Info] (c) 2018 - 2021 Viveris Technologies
Mar 18 09:04:25 steamdeck PluginLoader[5123]: [uMTPrd - 09:04:25 - Info] Sending command : addstorage:/run/media/mmcblk0p1 "SD Card" rw
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Add storage SD Card - Root Path: /run/media/mmcblk0p1 - UID: -1 - GID: -1 - Flags: 0x00000000
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Add storage Home - Root Path: /home/deck - UID: 1000 - GID: 1000 - Flags: 0x00000000
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Device path : /dev/ffs-mtp/ep0
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB In End point path : /dev/ffs-mtp/ep1
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Out End point path : /dev/ffs-mtp/ep2
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Event End point path : /dev/ffs-mtp/ep3
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Max packet size : 0x200 bytes
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Max write buffer size : 0x2000 bytes
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Max read buffer size : 0x2000 bytes
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Read file buffer size : 0x100000 bytes
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Manufacturer string : Valve
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Product string : Steam Deck
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Serial string : FVAA15101FDB
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Firmware Version string : Rev A
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Interface string : MTP
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Vendor ID : 0x28DE
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Product ID : 0x3000
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB class ID : 0x06
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB subclass ID : 0x01
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Protocol ID : 0x01
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB Device version : 0x3008
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] USB FunctionFS Mode
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Wait for connection : 0
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Loop on disconnect : 0
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Show hidden files : 1
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] File creation umask : System default umask
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Default UID : 1000
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] Default GID : 1000
Mar 18 09:04:25 steamdeck PluginLoader[5122]: [uMTPrd - 09:04:25 - Info] inotify : yes
Mar 18 09:04:25 steamdeck kernel: read descriptors
Mar 18 09:04:25 steamdeck kernel: read strings
```

Connected the usb cable:
`Mar 18 09:05:05 steamdeck PluginLoader[5122]: [uMTPrd - 09:05:05 - Info] uMTP Responder : Disconnected`

Went out and back in the DeckMTP plugin page and discovered switch was off (nothing happened in journalctl)
So i pressed the switch again:
```
Mar 18 09:06:30 steamdeck PluginLoader[5241]: mkdir: cannot create directory ‘cfg’: File exists
Mar 18 09:06:30 steamdeck PluginLoader[5242]: mount: /home/deck/homebrew/plugins/DeckMTP/bin/cfg: none already mounted on /home/deck/homebrew/plugins/DeckMTP/bin/cfg.
Mar 18 09:06:30 steamdeck PluginLoader[5242]:        dmesg(1) may have more information after failed mount system call.
Mar 18 09:06:30 steamdeck PluginLoader[5243]: mkdir: cannot create directory ‘cfg/usb_gadget/g1’: File exists
Mar 18 09:06:30 steamdeck PluginLoader[5244]: mkdir: cannot create directory ‘configs/c.1’: File exists
Mar 18 09:06:30 steamdeck PluginLoader[5245]: mkdir: cannot create directory ‘strings/0x409’: File exists
Mar 18 09:06:30 steamdeck PluginLoader[5246]: mkdir: cannot create directory ‘configs/c.1/strings/0x409’: File exists
Mar 18 09:06:30 steamdeck PluginLoader[5247]: mkdir: cannot create directory ‘functions/ffs.mtp’: File exists
Mar 18 09:06:30 steamdeck PluginLoader[5248]: ln: failed to create symbolic link 'configs/c.1/ffs.mtp': File exists
Mar 18 09:06:30 steamdeck PluginLoader[5249]: mkdir: cannot create directory ‘/dev/ffs-mtp’: File exists
[... cut ...]
Mar 18 09:06:30 steamdeck PluginLoader[5255]: [uMTPrd - 09:06:30 - Info] Default UID : 1000
Mar 18 09:06:30 steamdeck PluginLoader[5255]: [uMTPrd - 09:06:30 - Info] Default GID : 1000
Mar 18 09:06:30 steamdeck PluginLoader[5255]: [uMTPrd - 09:06:30 - Info] inotify : yes
Mar 18 09:06:30 steamdeck kernel: read descriptors
Mar 18 09:06:30 steamdeck kernel: read strings
```

After 40 seconds (cable plugged in since the first time I plugged it in):
`Mar 18 09:07:10 steamdeck PluginLoader[5255]: [uMTPrd - 09:07:10 - Info] uMTP Responder : Disconnected`

dafta, 2023-03-18:
Would you mind doing this again, and sending the whole log since boot and until the end? There is some very important information emitted here, and also, the line `Mar 17 21:59:56 steamdeck PluginLoader[3111]: ./stop.sh: line 5: echo: write error: No such device` is missing from the last logs you sent, but it was present in the earlier ones. This line is key, because if it is present every time, I know that it's got something to do with the UDC, which is a problem with certain OS and BIOS version combinations. The earlier messages are also important so that I can compare them with a working Deck.

cchrkk, 2023-03-18:
> Would you mind doing this again, and sending the whole log since boot and until the end? There is some very important information emitted here, and also, the line `Mar 17 21:59:56 steamdeck [...]

You can see the error appearing the second time I toggle the switch.
 Btw rebooted and this Is  the log from start to end:

`Last login: Sat Mar 18 17:26:48 2023 from 10.0.0.19
(deck@steamdeck ~)$ journalctl -f
Mar 18 17:29:52 steamdeck gpu-trace[966]: INFO - Executing get tracing status command
Mar 18 17:29:52 steamdeck gpu-trace[966]: 127.0.0.1 - - [18/Mar/2023 17:29:52] "POST / HTTP/1.1" 200 -
Mar 18 17:29:52 steamdeck gpu-trace[966]: INFO - Executing get tracing status command
Mar 18 17:29:52 steamdeck gpu-trace[966]: 127.0.0.1 - - [18/Mar/2023 17:29:52] "POST / HTTP/1.1" 200 -
Mar 18 17:29:52 steamdeck gpu-trace[966]: INFO - Executing get tracing status command
Mar 18 17:29:52 steamdeck gpu-trace[966]: 127.0.0.1 - - [18/Mar/2023 17:29:52] "POST / HTTP/1.1" 200 -
Mar 18 17:29:52 steamdeck pkexec[1862]: pam_unix(polkit-1:session): session opened for user root(uid=0) by (uid=1000)
Mar 18 17:29:52 steamdeck pkexec[1862]: deck: Executing command [USER=root] [TTY=unknown] [CWD=/home/deck/.local/share/Steam] [COMMAND=/usr/bin/steamos-polkit-helpers/steamos-priv-write /sys/class/drm/card0/device/power_dpm_force_performance_level auto]
Mar 18 17:29:52 steamdeck p-steamos-priv-write[1869]: checking: /sys/class/drm/card0/device/power_dpm_force_performance_level
Mar 18 17:29:52 steamdeck p-steamos-priv-write[1871]: commit: auto -> /sys/class/drm/card0/device/power_dpm_force_performance_level
Mar 18 17:29:52 steamdeck pkexec[1876]: pam_unix(polkit-1:session): session opened for user root(uid=0) by (uid=1000)
Mar 18 17:29:52 steamdeck pkexec[1876]: deck: Executing command [USER=root] [TTY=unknown] [CWD=/home/deck/.local/share/Steam] [COMMAND=/usr/bin/steamos-polkit-helpers/steamos-priv-write /sys/class/backlight/amdgpu_bl0/brightness 32767]
Mar 18 17:29:52 steamdeck p-steamos-priv-write[1884]: checking: /sys/class/backlight/amdgpu_bl0/brightness
Mar 18 17:29:52 steamdeck p-steamos-priv-write[1886]: commit: 32767 -> /sys/class/backlight/amdgpu_bl0/brightness
Mar 18 17:29:52 steamdeck rfkill[1894]: unblock set for type bluetooth
Mar 18 17:29:52 steamdeck kernel: input: Steam Deck as /devices/pci0000:00/0000:00:08.1/0000:04:00.4/usb3/3-3/3-3:1.2/0003:28DE:1205.0004/input/input24
Mar 18 17:29:53 steamdeck pkexec[1900]: pam_unix(polkit-1:session): session opened for user root(uid=0) by (uid=1000)
Mar 18 17:29:53 steamdeck pkexec[1900]: deck: Executing command [USER=root] [TTY=unknown] [CWD=/home/deck/.local/share/Steam] [COMMAND=/usr/bin/steamos-polkit-helpers/steamos-priv-write /sys/class/hwmon/hwmon4/power1_cap 15000000]
Mar 18 17:29:53 steamdeck p-steamos-priv-write[1924]: checking: /sys/class/hwmon/hwmon4/power1_cap
Mar 18 17:29:53 steamdeck p-steamos-priv-write[1926]: commit: 15000000 -> /sys/class/hwmon/hwmon4/power1_cap
Mar 18 17:29:53 steamdeck pkexec[1930]: pam_unix(polkit-1:session): session opened for user root(uid=0) by (uid=1000)
Mar 18 17:29:53 steamdeck pkexec[1930]: deck: Executing command [USER=root] [TTY=unknown] [CWD=/home/deck/.local/share/Steam] [COMMAND=/usr/bin/steamos-polkit-helpers/steamos-priv-write /sys/class/hwmon/hwmon4/power2_cap 15000000]
Mar 18 17:29:53 steamdeck p-steamos-priv-write[1937]: checking: /sys/class/hwmon/hwmon4/power2_cap
Mar 18 17:29:53 steamdeck p-steamos-priv-write[1939]: commit: 15000000 -> /sys/class/hwmon/hwmon4/power2_cap
Mar 18 17:29:53 steamdeck rtkit-daemon[1397]: Successfully made thread 1667 of process 1667 owned by '1000' high priority at nice level -10.
Mar 18 17:29:53 steamdeck rtkit-daemon[1397]: Supervising 9 threads of 4 processes of 1 users.
Mar 18 17:29:53 steamdeck pipewire-pulse[1839]: mod.echo-cancel: aec.method is not supported anymore use library.name
Mar 18 17:29:53 steamdeck sshd[1561]: Connection closed by authenticating user deck 10.0.0.19 port 45438 [preauth]
Mar 18 17:29:53 steamdeck PluginLoader[1383]: [css_utils][INFO]: [CSS_Loader] Result failed! Tab not found by lambda (['SP|Steam Big Picture Mode'])
Mar 18 17:29:53 steamdeck PluginLoader[1383]: [css_utils][INFO]: [CSS_Loader] Result failed! Tab not found by lambda (['SP|Steam Big Picture Mode'])
Mar 18 17:29:54 steamdeck kernel: input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input25
Mar 18 17:29:54 steamdeck PluginLoader[1383]: [css_utils][INFO]: [CSS_Loader] Result failed! Tab not found by lambda (['SP|Steam Big Picture Mode'])
Mar 18 17:29:54 steamdeck PluginLoader[1383]: [css_utils][INFO]: [CSS_Loader] Result failed! Tab not found by lambda (['SP|Steam Big Picture Mode'])
Mar 18 17:29:54 steamdeck dbus-daemon[964]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.52' (uid=1000 pid=2043 comm="timedatectl status")
Mar 18 17:29:54 steamdeck systemd[1]: Starting Time & Date Service...
Mar 18 17:29:54 steamdeck dbus-daemon[964]: [system] Successfully activated service 'org.freedesktop.timedate1'
Mar 18 17:29:54 steamdeck systemd[1]: Started Time & Date Service.
Mar 18 17:29:55 steamdeck systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Mar 18 17:29:55 steamdeck pkexec[2056]: pam_unix(polkit-1:session): session opened for user root(uid=0) by (uid=1000)
[... cut ...]
Mar 18 17:31:16 steamdeck systemd[1]: Unmounting /esp...
Mar 18 17:31:16 steamdeck systemd[1]: steamos-boot.service: Deactivated successfully.
Mar 18 17:31:16 steamdeck systemd[1]: Stopped SteamOS Boot Registration.
Mar 18 17:31:16 steamdeck systemd[1]: Unmounting /efi...
Mar 18 17:31:16 steamdeck systemd[1]: esp.mount: Deactivated successfully.
Mar 18 17:31:16 steamdeck systemd[1]: Unmounted /esp.
Mar 18 17:31:16 steamdeck systemd[1]: efi.mount: Deactivated successfully.
Mar 18 17:31:16 steamdeck systemd[1]: Unmounted /efi.`

dafta, 2023-03-18:
Okay, so this is interesting, it's not what I expected. Are you on stable SteamOS? And what is your BIOS version? You can check in Settings > System

cchrkk, 2023-03-19:
> Okay, so this is interesting, it's not what I expected. Are you on stable SteamOS? And what is your BIOS version? You can check in Settings > System

Actually i'm on Main channel, OS version 3.5, Build 20230313.1001, Kernel version 6.1.12-valve2-1-neptune-61, Bios is F7A0113

Didn't work on Stable and Beta too, that's the reason I upgraded to main channel to try but no luck

dafta, 2023-03-19:
If you're on BIOS 113, did you actually toggle the DRD switch to on, or was it already on by default? Could you try toggling it off and on again? There's a bug in that version of BIOS where it says that it's on by default, but you have to toggle it for it to actually become enabled.

Also, once you do that, if it doesn't work, I'd suggest switching to Stable and trying the same thing again, as there is a possible issue in Main that makes it fail. I don't know if they fixed it already or not, since I don't have my Deck to test, but there was a problem with it for a while on Main a month or so ago.

gameboycjp, 2023-03-19:
Mine did, actually. I tried turning off, saving, then going back and
turning it on, no dice. I tried causing the crash and checking the logs
before you suggested it but nothing ever came up so I've been
procrastinating, and now it doesn't seem to crash anymore after flipping
that setting around. I'll try multiple cables and computers again, will let
you know how it goes.

gameboycjp, 2023-03-19:
Tried with a windows desktop and an a to c cable, and with a Linux laptop
and an a to c and c to c cable. Still no dice. My desktop doesn't have a
USB c port so I can't try it with that.

gameboycjp, 2023-03-20:
> @gameboycjp <https://github.com/gameboycjp> Are you on Stable?

Yeah. Should I try a beta?

dafta, 2023-03-20:
No, it should work on Stable. Which BIOS version are you on? You can see that in Settings > System.

gameboycjp, 2023-03-20:
Same as the other person.

cchrkk, 2023-03-20:
> Same as the other person.
> […](#)

Btw I've tried to go in BIOS and flip the switch some times and save but it's still not working. I will try to go back to Stable channel and see

dafta, 2023-03-20:
There might have been some changes. I can't test this myself right now, unfortunately, but I will as soon as possible, sometime this week.

dafta, 2023-03-30:
I've tried Stable with the new BIOS, so version 115. I'm not sure if that has yet reached Stable, but I upgraded it manually, and everything seems to work. I'm curios to see which BIOS version everyone here is on. A BIOS upgrade will probably fix the issues.

TowelSniffer, 2023-04-15:
I can confirm updating to bios 115 fixed the issue for me

cchrkk, 2023-04-15:
> I can confirm updating to bios 115 fixed the issue for me

Thank you, later today I will try with 115

dafta, 2023-05-04:
@cchrkk Did it work?

cchrkk, 2023-05-04:
> @cchrkk Did it work?

It did work perfectly BUT I don't know what happened and I hard to reimage my deck. Switched to stable, BIOS still 115 but It doesnt work anymore now🤷🏼‍♂️  Must be something on my end, no problem 😂

dafta, 2023-05-04:
Damn, too bad hahah.

Okay, closing the issue.

closed 2023-05-04 by dafta

gabrielmaster123, 2023-05-06:
I still have the same problem after updating to bios version F7A0115 by changing my branch to main and forcing the bioos update

dafta, 2023-05-06:
The main branch can have any number of new bugs introduced as it's a preview of the current state of development. If you wouldn't mind, switch back to stable, and if the BIOS gets downgraded, upgrade it to 115 manually. You can do so by downloading the 115 BIOS from [here](https://gitlab.com/evlaV/jupiter-hw-support/-/blob/master/usr/share/jupiter_bios/F7A0115_sign.fd), and using the following command:

```
sudo /usr/share/jupiter_bios_updater/h2offt F7A0115_sign.fd -ALL
```

gabrielmaster123, 2023-05-06:
That's what I did sorry if I didn't articulate myself well, I only used
main for the bios update

dafta, 2023-05-08:
If you're on stable now, could you check if your BIOS version is indeed 115, and that it didn't get downgraded, just in case?

gabrielmaster123, 2023-05-08:
It is

dafta, 2023-05-23:
Either desktop or gaming mode is fine, both is the same.

Is that the only line of output you get? There should be more. You're on Stable now, right?

gabrielmaster123, 2023-05-23:
> Stable now, right? Can you check?

Im on stable thats the only line I get

dafta, 2023-05-30:
Hmm, that's strange. On my Deck, Stable and BIOS 115,  I get the following:

```
dwc3                  172032  0
roles                  16384  1 dwc3
ulpi                   20480  1 dwc3
udc_core               77824  1 dwc3
dwc3_pci               24576  0
```

This is how it should look. You could potentially try rebooting the Deck, then the command `sudo modprobe dwc3` in desktop mode, and then switch back to game mode and test the plugin again. Other than that, maybe try it on a recovery image or do a hard reset on your Deck. Something might be misconfigured on your OS.