dafta/DeckMTP #4: [Feature Request] Expose mounted filesystems

Riven-Spell, 2023-03-30:
Currently, only the home directory is exposed, which is fine if you symlink other drives into the home directory, but it would be nice to have other filesystems (e.g. the MicroSD) exposed via MTP here

[image]

Riven-Spell, 2023-03-31:
Oh, maybe I'm experiencing a bug then. Oddly, it's not picking my SD up.

dafta, 2023-03-31:
It's possible that your Deck is mounting the SD card in a different location. I know that Valve did do something with the mount location of SD cards a while back that broke Emudeck. Could you please post the output of the `lsblk` command?

closed 2023-07-12 by dafta