dafta/DeckMTP #5: Deck cannot go to sleep mode when connected to PC

Honzackcz, 2023-04-06:
I don't really think that is DeckMTP bug but may occur for those who use it.

When Deck is connected to a PC and power button is pressed shortly to put Deck to sleep mode, display will turn off and sound for entering sleep mode is played. But after few seconds another sound is played, display is still off and Deck is now reacting to inputs by doing menu sounds. Another power button press will do the same.
In this state after Deck is disconnected from PC, it can be putted into proper sleep mode and then it will wake up normally.

This issue seems more like a USB DRD bug then DeckMTP because it doesn't matter if DeckMTP is enabled or not.

For me specifically it complicates a bit charging while I am at home because I use PC's front panel USB-C for it. (It have 60W PD)

dafta, 2023-05-04:
Huh, I haven't noticed this until now. I tried it and it really doesn't go to sleep, on BIOS 115. Unfortunately, this is something Valve has to fix, it's most likely a BIOS or driver issue.

dafta, 2024-03-05:
Should be fixed when the newest BIOS hits stable.

closed 2024-03-05 by dafta