dafta/DeckMTP #7: Not recognized with Mac OS

V3ntus, 2023-04-17:
I'm aware MTP doesn't seem to be natively supported in Mac OS, but I've tried OpenMTP and Android File Transfer, yet none of which recognize the Deck, although they have worked with my phone before. Is there a known way of doing this?

Mac OS 12.6.1 Intel
Deck BIOS ver: F7A0113

ZeroEcks, 2023-05-03:
Seconding that I tried "[MTP for Mac](https://www.hyperintegrate.com/products/mtp-for-mac)" and it doesn't work, but does with my phone. Maybe the mode of the USB port isn't being set right?

dafta, 2023-05-04:
Do you have another PC you could try it with? I don't think it's a problem with MTP, but rather, DRD not working correctly on your Deck.

That being said, there are some issues with DRD specifically on the 113 BIOS, which seem to be fixed with the 115 BIOS. I would recommend upgrading to the 115 BIOS.

You can download the 115 BIOS here: https://gitlab.com/evlaV/jupiter-hw-support/-/blob/master/usr/share/jupiter_bios/F7A0115_sign.fd

To upgrade the BIOS, you can use the following command:

```
sudo /usr/share/jupiter_bios_updater/h2offt F7A0115_sign.fd -ALL
```

closed 2023-05-30 by dafta