dafta/DeckMTP #8: Enable MTP doesn't work, rendering DeckMTP unusable

BrochachoTheBro, 2023-04-18:
DeckMTP has worked flawlessly before, but now it doesn't work for me. I go into the DeckMTP toggle in Decky to enable it, but my computer doesn't recognize it at all like it used to. Additionally, when I exit the DeckMTP section and go back into it, the toggle has switched back to being off. I can confirm I have DRD set up, and again, it used to work just fine.

parkerlreed, 2023-04-18:
If you are on BIOS 113 it's known broken. 115 is needed for working USB guest support.

BrochachoTheBro, 2023-04-18:
It looks like my BIOS version is F7A0110. Should I update it then? How could one go about doing so?
(Also sorry if what all I'm asking is rather noob-y in nature)

BrochachoTheBro, 2023-04-19:
Very odd. Idk if I did anything or not, but it felt so sudden. I don't think it's the cable, because I still get that slow charging warning like I always did, so it knows its plugged into something.
Should I just update my BIOS then or?

BrochachoTheBro, 2023-04-20:
Alright nevermind, reinstalling seems to have fixed the issue.

closed 2023-04-20 by BrochachoTheBro