dafta/DeckMTP #9: MTP driver crash on windows with usb 3.2

KoTeYkA23, 2023-05-16:
Hi, I want to report a problem with usb 3.2 driver crash on windows, I tried onboard usb as well as pci e usb card, doesnt matter usb a or c, I only manage get it randomly work with type C cable 1 time, but it stoped work after update as well. Usb 2.0 cables work 100% fine. MTP Driver return code 10 (Insufficient system resources to complete the API call) or something like that (sorry I not use english version windows so that might be not exact description in device manager). My android phone works fine on any cables I tested. My deck bios version is 110. My computer is: rysen 5900x, b550 chipset, win11.

Anyway thanks in advance guys for doing valves work

dafta, 2023-05-23:
This might be an issue with the driver installed on Windows. I haven't personally tested on Windows 11, might also be an issue.

You could try uninstalling the driver, plugging in the Deck again, and enabling DeckMTP, to let Windows reinstall the driver. If that doesn't work, we can see about other troubleshooting steps.

KoTeYkA23, 2023-05-24:
Nope, nothing change if I delete driver or updating it to similar, in the driver info driver date development is 2006 year, so it might be not only on win11

KoTeYkA23, 2023-05-30:
Nope, after fully shutting down pc and reboot steamdeck mtp driver still crash (c to c but it souldn't matter), but it worked with front usb a to c, also I would like to report other issue, may be I'm just dum at permission managment in linux but my games folder on main drive that use heroic just not show up in mtp, after creating other folder it shows up normally, but as soon as I add game on that folder in heroic(and wine prefixes too) it disappears from mtp, I tried to add all permission for my user but nothing changed, it also happening with other folders, but I have no explanation of them dissapearing, here is screenshot of ssh and mtp in same folder
[image]

KoTeYkA23, 2023-05-30:
Front usb was 2.0 btw, so it didn't change anything

TuNdRa-Plains, 2023-11-18:
This seems partially related to Windows in some form - Windows 11 exhibits this behaviour on X570 USB 3.x Controllers.
However the USB 2.0 headers will connect & Run MTP Mode fine. I've dug through most of the Windows 11 suggested fixes short of a complete OS Reinstall and nothing resolves it on USB3. (This is across: Direct headers on the board, rear-panel USB, 2 different USB3 Hubs, 2 different cables)

Just for dev ref; this is on SteamOS 3.5.5, Bios F7A0119.

Honzackcz, 2023-11-23:
> This seems partially related to Windows in some form - Windows 11 exhibits this behaviour on X570 USB 3.x Controllers. However the USB 2.0 headers will connect & Run MTP Mode fine. I've dug through [...]

I can add to this with Windows 11 and X670E, any faster USB connection then 2.0 will crash while transferring a file, it can delete files and create empty files tho. (SteamOS 3.5.5, BIOS F7A0119)

It is the same thing with this issue:
https://github.com/dafta/DeckMTP/issues/2#issuecomment-1485569340

Honzackcz, 2024-06-18:
DeckMTP 1.0.3 still doesn't work with higher USB version then 2.0 (SteamOS 3.5.19, BIOS F7A0120)

The USB-C to UBS-C issue ( https://github.com/dafta/DeckMTP/issues/2 ) is most likely the same issue with this one because only USB-C to USB-C cables I have are 3.0+

dafta, 2024-06-18:
Did you try it again just now? Because the 1.0.3 version on the testing store was just updated an hour ago. Did you update it?

Honzackcz, 2024-06-18:
Yep, I download the 1.0.3-f0917db from the testing store.

dafta, 2024-06-18:
Okay, thanks for the info. I will try to replicate this and see if I can't find the issue.

By the way, I noticed you are using the F7A0120 BIOS version, which still had some issues with DRD. Would you be willing to test with version F7A0121 or newer?

Honzackcz, 2024-06-18:
Well, I don't know how to update the BIOS manually. If it isn't too complicated, I may try.

dafta, 2024-06-18:
It's not too complicated, it's just one command. You can download either the 121 update from [here](https://gitlab.com/evlaV/jupiter-hw-support/-/raw/7ffc22a4dc083c005e26676d276bdbd90dd1de5e/usr/share/jupiter_bios/F7A0121_sign.fd), or the newer 131 update from [here](https://gitlab.com/evlaV/jupiter-hw-support/-/raw/master/usr/share/jupiter_bios/F7A0131_sign.fd). This is if you are on the LCD deck. For the OLED decks, there is a different BIOS.

And then use the following command to update it:

```
sudo /usr/share/jupiter_bios_updater/h2offt F7A0121_sign.fd -all -AC
```

You would change the `F7A0121_sign.fd` to `F7A0131_sign.fd` if you downloaded the 131 BIOS.

This is how the deck updates it as well, it's the "official" method of updating the BIOS, but I know updating the BIOS manually can be scary, and you're not obligated to do it.

Honzackcz, 2024-06-18:
OK, I tried the F7A0131 and it still doesn't work.

dafta, 2024-06-18:
This is on Windows?

Honzackcz, 2024-06-18:
OH, I figured something out. Some time ago I did a bit of research on how exactly are connected USBs on my motherboard. There I find out that most of the USB controllers are from AMD but one that is used for 40Gbps USBs is from Intel. So, now I tried to use the Intel controller USB and it works just fine.

[image]

dafta, 2024-06-18:
Hmm okay, that's interesting. It could be some kind of hardware issue. Explains why I could never reproduce it, too. I'll see if I can find some machine where this happens, I could maybe detect it and lower the speed in that case.

Is this PC running Windows? Because I've seen reports that it might also be a Windows driver issue.

Honzackcz, 2024-06-18:
Yep, Windows 11

dafta, 2024-06-18:
As far as I can tell, all of the people that have this issue are on Windows 11. If anyone has this issue on Windows 10, can you please notify me here?

Also if anyone who has this issue can test on Windows 10 on the same machine, that would be amazing.

KoTeYkA23, 2024-06-18:
Hi there, I tried build from main branch and it didnt work, only thing is message to enable drd when it disabled in bios, its not working nor in Linux nor in Win10, in linux dmesg reporting

```
config error 
Cannot enable. Maybe the USB cable is bad?
```

I tried other cables and ports but result is not any better
my deck bios version is F7A0120

dafta, 2024-06-18:
Did you install from the decky store? The 1.0.3 version from testing?

It could also be an issue with the 120 BIOS. You could try disabling DRD, restarting the deck, then enabling DRD again, or upgrading the BIOS as explained in the comment [here](https://github.com/dafta/DeckMTP/issues/9#issuecomment-2176622297).

KoTeYkA23, 2024-06-18:
> Did you install from the decky store? The 1.0.3 version from testing?

No, I didnt look up for that, I builded from main, I'll give it a try if I manage to find that version

KoTeYkA23, 2024-06-18:
Interesting, last version from testing made my deck detect as a camera in Linux, but no files are available, discover either drop me in my home folder or in that nonsense
`camera:/USB%2520PTP%2520Class%2520Camera@usb:006,007/`

KoTeYkA23, 2024-06-18:
Oh, it actually works, but again, only on usb 2

dafta, 2024-06-18:
@KoTeYkA23 You had this issue with a Linux PC?

I did some testing and I also found some ports that don't work on my PC. Since this happens on Linux as well, I think that rules out driver issues on the PC.

I did some testing with other USB gadgets as well, besides MTP, and they don't work on those ports either.

At this point I'm thinking it's either a hardware issue, an issue with the USB driver on the deck, or an issue with the deck BIOS. I will definitely have to do more testing tomorrow, though. It might be that these ports didn't work for other reasons, and I haven't tried a USB 2 cable with them, yet. Will do that tomorrow.

KoTeYkA23, 2024-06-18:
> @KoTeYkA23 You had this issue with a Linux PC?

Yeah, I moved to Linux with win10 dulabooting, so its the same pc that I created issue about

> At this point I'm thinking it's either a hardware issue, an issue with the USB driver on the deck

Seems very likely, cause one of my port dont even ''slocharge" deck, but works completly fine with my android phone and mtp on it

dafta, 2024-06-18:
> Seems very likely, cause one of my port dont even ''slowcharge" deck, but works completly fine with my android phone and mtp on it

If you turn off DRD in BIOS and go back to xHCI, does the deck slowcharge then? If so, I think we can at least isolate this issue to either the BIOS or the USB drivers.

KoTeYkA23, 2024-06-19:
> If you turn off DRD in BIOS and go back to xHCI, does the deck slowcharge then? If so, I think we can at least isolate this issue to either the BIOS or the USB drivers.
Yep, turned drd off and it still doesnt charge, so it either hardware problem or deck's software

drublic123, 2024-07-14:
Steam Deck OLED
OS - 3.5.19
Bios - F7G0107
DeckMTP 1.0.3-f0917db

PC
Windows 11
AM4 system

USB2.0 works but with USB 3.2 it connects but any data transfer will close the MTP connection all together.

Used the USB Device Tree Viewer and found this error when it was connected to USB 3.2, any chance the value can be updated?

wMaxPacketSize           : 0x0200 (max 512 bytes)  *!*ERROR  SuperSpeed Bulk endpoints must be 1024 bytes

to 

wMaxPacketSize           : 0x0400 (max 1024 bytes)

dafta, 2024-07-14:
> Used the USB Device Tree Viewer and found this error when it was connected to USB 3.0, any chance the value can be updated?

Thank you, this is very helpful. I'll update the value and push a new build to the testing store.

drublic123, 2024-07-17:
> > Used the USB Device Tree Viewer and found this error when it was connected to USB 3.0, any chance the value can be updated?
> 
> Thank you, this is very helpful. I'll update the value and push a new build to the testing store.

Any update to testing version?

dafta, 2024-07-17:
> Any update to testing version?

Sorry, I don't have access to my dev machine for the time being, and I'd like to test this change for any potential problems before I push it as default.

However, you can try this for yourself already fairly easily. You'd just have to change [line 114 in config file umtprd.conf here](https://github.com/dafta/DeckMTP/blob/3ff8b6522417849c2e21d11e59ce9968445e617b/backend/src/umtprd.conf#L114) to say `usb_max_packet_size 0x400`. On the Steam Deck, this file is available in your home directory, under `~/homebrew/plugins/DeckMTP/bin/umtprd.conf`. If you do, please let me know if this fixes your issue or introduces new issues.

drublic123, 2024-07-18:
Ok great updated the values to this usb_max_packet_size 0x400 and my front USB 3.0 ports now work fine without data drop outs. Transferred 27.5gb in 2mnin 52sec.

USB 2.0 ports won't work now (not a big deal).

The rear USB 3.0 ports makes the detection noise but wont pop up the device, the rear ports must be a different USB controller?

**Front USB 3.0 port (USB Tree View)**
Vendor ID                : 0x28DE (Valve Corporation)
Product ID               : 0x3000
USB Version              : 3.2 Gen 2
Port maximum Speed       : SuperSpeed
Device maximum Speed     : SuperSpeedPlus or higher
Device Connection Speed  : SuperSpeed
Self powered             : no
Demanded Current         : 120 mA
Used Endpoints           : 4

**Rear USB 3.0 port (USB Tree View)**
Vendor ID                : 0x28DE (Valve Corporation)
Product ID               : 0x3000
USB Version              : 3.2 Gen 2
Port maximum Speed       : SuperSpeedPlus or higher
Device maximum Speed     : SuperSpeedPlus or higher
Device Connection Speed  : SuperSpeedPlus or higher
Self powered             : no
Demanded Current         : 120 mA
**Device Manager Problem   : 10 (CM_PROB_FAILED_START)**
Used Endpoints           : 1

drublic123, 2024-07-18:
----------------- Endpoint Descriptor -----------------
bLength                  : 0x07 (7 bytes)
bDescriptorType          : 0x05 (Endpoint Descriptor)
bEndpointAddress         : 0x82 (Direction=IN EndpointID=2)
bmAttributes             : 0x03 (TransferType=InterruptPeriodic)
**wMaxPacketSize           : 0x001C  *!*ERROR  SuperSpeed interrupt endpoints must have wMaxPacketSize value of 1024 bytes when the SuperSpeed endpoint companion descriptor bMaxBurst value is >0**
bInterval                : 0x06 (32 microframes -> 4 ms)
Data (HexDump)           : 07 05 82 03 1C 00 06   

Probably another area in the code  that needs to be updated

dafta, 2024-08-04:
Given that this config change works for USB 3.2 but stops USB 2.0 from working, this change will probably have to wait for the next version, because I'll probably have to add settings to change this manually. Unfortunately I can't programatically detect the speed of the connected port when the Deck is in device mode, or at least I still haven't found a way.