dafta/DeckMTP #10: Does not seem to work suddenly?

123ika, 2023-06-03:
Hi, this plugin does not seem to want to work anymore on my Deck. I've turned it on, plugged in the same cable I've been using so far and that I know worked before and it does not pop up the Deck on my Windows. I've even tried a different cable. Any ideas?

OS Update Channel: Beta
Steam Client Update Channel: Beta
Decky Loader Update Channel: Prelease
Store Channel: Default
DeckMTP 1.0.2

lividhen, 2023-06-06:
Also experiencing this. The modprobe storage gadget also seems to be broken.
os channel stable
steam client beta
decky loader prerelease

dafta, 2023-06-06:
What BIOS version are you on? You can check that in Settings > System.

If it's 113 or 114, these BIOS versions have known issues with USB Dual-Role Device. I'd recommend updating your BIOS to 115. This is the command:

    sudo /usr/share/jupiter_bios_updater/h2offt F7A0115_sign.fd -ALL

The BIOS file has to be in the same folder that you run the command in, of course (or add the path before the filename).

You can get the BIOS which definitely works from here: https://gitlab.com/evlaV/jupiter-hw-support/-/blob/5644a5692db16b429b09e48e278b484a2d1d4602/usr/share/jupiter_bios/F7A0115_sign.fd

lividhen, 2023-06-06:
Yep! Was on 113. Thanks! I was unaware there was a bug.

123ika, 2023-06-07:
> What BIOS version are you on? You can check that in Settings > System.
>
> If it's 113 or 114, these BIOS versions have known issues with USB Dual-Role Device. I'd recommend updating your BIOS to 115. [...]

I'm on F7A0116 and I'm still getting the issue, I did use the Steam Deck "Main" OS Update Channel a while back to go to OS Version 3.5 before reverting back, looks like the BIOS version from there stuck, is it safe to downgrade or should I wait?

parkerlreed, 2023-06-07:
Downgrades work fine in my experience.

dafta, 2023-06-08:
Downgrades should work, and do work from my experience, but just in case, it's good to reset the BIOS settings beforehand. This is what a Valve employee told me once:

Just a note, while it generally works to manually downgrade between recent BIOS versions, this isn't fully tested/supported. If you are trying it, it is always a good idea to reset BIOS settings when doing something like this: Power the Deck down, then power it up whilst holding the ... and Volume+ keys. (The display will take 10s+ to wake up when doing this and the LED will flash at you)

123ika, 2023-06-17:
Can confirm downgrading did the trick! Thanks :)

closed 2023-06-17 by 123ika

whitewolf185, 2023-08-10:
Hello!
I have 3.4.8v OC and 115 bios. 
Windows writes that it cannot detect device (see screen shot below)

[image]

dafta, 2023-08-14:
This could be due to the Windows drivers. Do you perhaps have another device to test this on?

whitewolf185, 2023-08-14:
> This could be due to the Windows drivers. Do you perhaps have another device to test this on?

Like usb flash drive? Do not clearly understand you.

Actually, reload works for me. However, if i will disconnect my device, the problem comes back

ergoz, 2023-11-24:
> Hello! I have 3.4.8v OC and 115 bios. Windows writes that it cannot detect device (see screen shot below) [...]

same problem

ergoz, 2023-11-24:
i found something interesting.

1. I enable DeckMTP, exit from page where toggle. 
2. Connect cable to PC (or macbook, it has same problem).
3. After 5 seconds i open DeckMTP plugin page and it has disabled status.

I think it tries to run when connecting to pc, got error and crushes... that is why it changes status from enabled to disabled by itself

Sorry for my English, it is not native for me

WhoTheFlower, 2023-12-05:
The Command if your path is set
sudo /usr/share/jupiter_bios_updater/h2offt F7A0115_sign.fd -ALL

The Command if your path is not set
sudo /usr/share/jupiter_bios_updater/h2offt ~/Downloads/F7A0115_sign.fd -ALL"

Credit to Dafta