dafta/DeckMTP #11: unable to detect custom name sdcard

Merith-TK, 2023-06-10:
SO, my sdcard is named `gamedrive`, which reflects in its mount path
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0     8G  0 loop /home/.steamos/offload/opt/rwfus/mount
                                      /opt/rwfus/mount
mmcblk0     179:0    0 238.8G  0 disk
└─mmcblk0p1 179:1    0 238.7G  0 part /run/media/gamedrive
nvme0n1     259:0    0  57.6G  0 disk
├─nvme0n1p1 259:1    0    64M  0 part
├─nvme0n1p2 259:2    0    32M  0 part
├─nvme0n1p3 259:3    0    32M  0 part
├─nvme0n1p4 259:4    0     5G  0 part /
├─nvme0n1p5 259:5    0     5G  0 part
├─nvme0n1p6 259:6    0   256M  0 part /var
├─nvme0n1p7 259:7    0   256M  0 part
└─nvme0n1p8 259:8    0    47G  0 part /usr/local
                                      /var/tmp
                                      /var/log
                                      /var/lib/systemd/coredump
                                      /var/lib/flatpak
                                      /var/lib/docker
                                      /srv
                                      /root
                                      /opt
                                      /home
```

That is where it mounts, because that is the name assigned to the filesystem

possible solution is to get mouth path based off off using `grep`
```
(deck@steamdeck ~)$ mount | grep mmcblk
/dev/mmcblk0p1 on /run/media/gamedrive type ext4 (rw,noatime)
```

dafta, 2023-07-12:
I'll have a fix for this, in the meantime, you could change the config file for uMTP, in the plugin's bin directory. The file is named `umtprd
conf`, and on line 26 yoi have an example of a share. You could add a line for your sd card there.

Merith-TK, 2023-07-12:
I mean I have an PR that does this without issue #12

commit 0c984cec (2024-03-05): Add support for multiple SD card partitions and custom names

closes #11
closed 2024-03-05 by dafta