dafta/DeckMTP #14: MTP mode doesn't work

x1unix, 2023-07-25:
I tried to install DeckMTP plugin and connect my Deck to Windows PC.
Also, I enabled dual role device mode in BIOS settings as mentioned in the README.

I used a pair of cables which 100% work. I often use them for my Nintendo switch for MTP and RCM.
Those are USB-C to USB-A and USB-C to USB-C Google Pixel cables.

It didn't work, so I checked `dmesg` for some messages:

```
[  202.351102] ffs_data_put(): freeing
[  204.551147] jupiter VLV0100:00: ACPI event [0x80], PDCS = 0
[  204.551179] jupiter VLV0100:00: jupiter_usb_role_work PDCS = 0
[  204.551183] jupiter VLV0100:00: USB role is device
[  231.867588] file system registered
[  231.881290] read descriptors
[  231.881298] read strings
[  234.643818] unloading
[  234.674820] ffs_data_put(): freeing
[  237.152504] jupiter VLV0100:00: ACPI event [0x80], PDCS = 21
[  237.486282] usb usb2-port1: config error
[  239.255664] jupiter VLV0100:00: jupiter_usb_role_work PDCS = 21
[  239.255674] jupiter VLV0100:00: USB role is device
[  239.855877] file system registered
[  239.871826] read descriptors
[  239.871834] read strings
[  241.538834] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[  241.538856] usb usb2-port1: config error
[  245.598783] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[  245.598809] usb usb2-port1: config error
[  249.062658] unloading
[  249.088457] ffs_data_put(): freeing
[  249.668799] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[  249.668825] usb usb2-port1: config error
[  253.159940] file system registered
[  253.173445] read descriptors
[  253.173452] read strings
[  253.731882] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
[  253.731908] usb usb2-port1: config error
[  257.798457] usb usb2-port1: Cannot enable. Maybe the USB cable is bad?
````

Log complains about the cable. I tried multiple cables that 100% work but get the same error.

parkerlreed, 2023-07-25:
What BIOS are you on?

x1unix, 2023-07-26:
@parkerlreed BIOS version is `F7A0113`

[image]

[image]

parkerlreed, 2023-07-26:
@x1unix 113 is broken for dual role as well. You want 115

https://gitlab.com/evlaV/jupiter-hw-support/-/raw/5644a5692db16b429b09e48e278b484a2d1d4602/usr/share/jupiter_bios/F7A0115_sign.fd?inline=false

x1unix, 2023-07-26:
@parkerlreed thanks. How do I update bios firmware?

parkerlreed, 2023-07-27:
> @parkerlreed thanks. How do I update bios firmware?

```
sudo /usr/share/jupiter_bios_updater/h2offt F7A0115_sign.fd -all -AC
```

MegaBaconSticks, 2023-07-28:
Just did this for myself and all of the boot options just disappeared. Tread carefully 🫡

dafta, 2023-08-02:
If your boot options disappeared, you might want to try resetting the BIOS settings like this: Power the Deck down, then power it up whilst holding the ... and Volume+ keys. (The display will take 10s+ to wake up when doing this and the LED will flash at you)

duirron, 2023-08-03:
> @x1unix 113 is broken for dual role as well. You want 115
> 
> https://gitlab.com/evlaV/jupiter-hw-support/-/raw/5644a5692db16b429b09e48e278b484a2d1d4602/usr/share/jupiter_bios/F7A0115_sign.fd?inline=false

Nice, this is the solution. it worked for me

duirron, 2023-08-03:
> > @parkerlreed thanks. How do I update bios firmware?
> 
> ```
> sudo /usr/share/jupiter_bios_updater/h2offt F7A0115_sign.fd -all -AC
> ```

This too

x1unix, 2023-08-13:
Updated the bios, this still doesn't work
[image]

dafta, 2023-08-14:
This could be due to the Windows drivers. Do you perhaps have another device to test this on?

megapro17, 2023-08-23:
I had 114 uefi and DRD enabled from factory, and it didn't worked even after updating to latest 116. Disabling and enabling it helped (I clicked save and exit twice)

parkerlreed, 2023-08-23:
> I had 114 uefi and DRD enabled from factory, and it didn't worked even after updating to latest 116. Disabling and enabling it helped (I clicked save and exit twice)

Again 115 is the *ONLY* recent version known to work. Everything else has DRD broken.

megapro17, 2023-08-23:
It's not broken, it required disabling and enabling that

parkerlreed, 2023-08-23:
Even then the enable and disable doesn't work on some versions. Especially 116.

JV-root, 2023-10-20:
Hello, Guys! My firmware version is 116, and MTP isn't working; the Steam Deck doesn't even show up on Windows. I'd really like to use it to make transferring some files easier. What are the risks of downgrading to 115? @parkerlreed

megapro17, 2023-10-20:
You should enable drd save, enable xchi save, and enable drd save

dafta, 2024-03-05:
DRD is broken for now, until the newest BIOS hits stable, at least.

closed 2024-03-05 by dafta

megapro17, 2024-03-08:
@dafta It does work