dafta/DeckMTP #16: DeckMTP does't work with SteamOS 3.4.10

wzchwzch, 2023-09-08:
DeckMTP does't work when I update SteamOS from 3.4.8 to 3.4.10

anhcestor, 2023-09-15:
Same here doesn’t work either

parkerlreed, 2023-09-20:
You need to be on BIOS 115 or 118 from the current Main channel. 116 dual role does not work

anhcestor, 2023-09-20:
> You need to be on BIOS 115 or 118 from the current Main channel. 116 dual role does not work

I am on bios 115 and it doesn’t work it has always worked for me for months should I upgrade to 118?

parkerlreed, 2023-09-20:
115 should be working. Is it giving you any errors?

anhcestor, 2023-09-20:
> 115 should be working. Is it giving you any errors?

It is my windows laptop can’t read any of it and also for reference I am on windows 10 has always worked fine for me for months on bios 115

anhcestor, 2023-09-27:
> DeckMTP does't work when I update SteamOS from 3.4.8 to 3.4.10

I finally found a fix first download this file https://gitlab.com/evlaV/jupiter-hw-support/-/blob/5644a5692db16b429b09e48e278b484a2d1d4602/usr/share/jupiter_bios/F7A0115_sign.fd

After downloading this file go to downloads in Dolphin and right-click the file to open in the terminal. After the file is opened in terminal. Paste this command below 

sudo /usr/share/jupiter_bios_updater/h2offt F7A0115_sign.fd -ALL

After pasting the command it will ask for your sudo password type it in and after that your steam deck will start downloading the bios and you should be good. I was already on bios 115 but had to redownload and this was the only thing that worked.

parkerlreed, 2023-09-27:
Dual Role Device works on 118 for what it's worth.

anhcestor, 2023-09-27:
> Dual Role Device works on 118 for what it's worth.

Should I just upgrade to 118 then ? Also where can I find the bios 118 file.

parkerlreed, 2023-09-29:
118 can be had from either Main channel or via the same repo linked above. 

Keep in mind the BIOS unlocker doesnt work for 118 if you were doing over/under clocking.

Ditpo, 2023-10-02:
Just a heads up: updating to 118 broke audio and touch on my deck, reverting to 115 fixed that.

Don't know if it's an issue with mainboard revisions, I bought one of the first valve refurbished units when they started selling them.

parkerlreed, 2023-10-02:
> Just a heads up: updating to 118 broke audio and touch on my deck, reverting to 115 fixed that. Steam Deck OS 3.4.10
> 
> Don't know if it's an issue with mainboard revisions, I bought one of the first valve refurbished units when they started selling them.

118 will break audio and touch if you "Load optimized defaults"

Should be fine otherwise.

Ditpo, 2023-10-02:
> 118 will break audio and touch if you "Load optimized defaults"
> 
> Should be fine otherwise.

Any way to fix that afterwards without flashing an older bios? If not, then that's not really a solution that should be offered imo.

There's always a chance that sometime down the line one might load optimized defaults and not know what happened...

parkerlreed, 2023-10-03:
Which is why 118 is still in testing and not released to beta or stable :)

You can upgrade to 118 and it will be fine as long as you don't load the defaults. 

Ideally 118 isn't what is going to hit stable in the future. Probably some later revision in that case.

L1Z3, 2023-10-22:
Doesn't seem like BIOS 118 works for me.

gezawatt, 2023-11-04:
I'm in bios 110 right now, tested bios versions 113, 115, 116, 118 and 119... On all of them, MTP didn't work, so I'm staying on bios 110 for now..

TheCyberQuake, 2023-11-18:
Did Bios 119 break it again? Because I just got updated from 116 to 119, DeckMTP still not working

dafta, 2024-03-05:
DRD is broken for now, at least until the newest BIOS hits stable.

closed 2024-03-05 by dafta