dafta/DeckMTP #17: Compatibility Issue - Steam Deck V3.5.7

Archer201977, 2023-11-24:
Hi, DeckMTP is awesome, which does what it says it does, Thank you! However, I found that it is not anymore working right after i updated my steam deck to V3.5.7 which in turn prevents DeckMTP from working, Is there a plan for updating the software you created or maybe you won't be doing it in the future? Thanks a lot for creating and easy to a use app!

Honzackcz, 2023-11-25:
Same issue here, now with SteamOS 3.5.7 DeckMTP doesn't work. I it did before on 3.5.5. Both with BIOS F7A0119.

Archer201977, 2023-11-25:
Well i figured i rollback to 3.5.5 and its back to working, for now i will just install what i need and then after that go and update to the latest version, hopefully the dev will update the plugin, its very much convenient faster with no breaks, better than wireless transfers.

FedorLvov, 2023-11-27:
I have same issue on 3.5.7. Rolling back to 3.5.5 helps, but only partially: the connection is interrupted after a few minutes, and I can’t transfer large files because of this.

Archer201977, 2023-11-27:
> I have same issue on 3.5.7. Rolling back to 3.5.5 helps, but only partially: the connection is interrupted after a few minutes, and I can’t transfer large files because of this.

Odd.. Large files transfers is very much fine with me, interruption is caused by the steam deck's function, just disabled the sleep when connected option, that will get rid of the interruption

11Nate11, 2023-11-30:
Couldn't get it to work even by rolling back to 3.5.5 any tips?

11Nate11, 2023-11-30:
> > Couldn't get it to work even by rolling back to 3.5.5 any tips?
> 
> Are you sure when you rollback and restarted its still on 3.5.5, because when you rollback to 3.5.5 and then restarted it reverts back to 3.5.7 unless if you add this in the terminal "sudo rauc status mark-active booted" after rolling back. I have no issues on deckmtp under 3.5.5 right after i rolledback and still its working properly even up today.

Yeah I made sure I was in 3.5.5, I uninstalled decky, wiped all the files and then reinstalled. I'm using the oled model and it came with drd already enabled in the bios so I don't know if that has anything to do with it. Thanks for your help though! I'll be waiting for the update:)

Archer201977, 2023-11-30:
> Yeah I made sure I was in 3.5.5, I uninstalled decky, wiped all the files and then reinstalled. I'm using the oled model and it came with drd already enabled in the bios so I don't know if that has [...]

I guess it may have something to do with your model since you got the Oled version, the chipset are different than the OG version.

gittyno1, 2023-12-05:
> I will be doing an update sometime soon, as soon as I identify the issue. Most likely an issue with MTP over USB 3 on Windows, I might add a Windows compatibility mode as a workaround for now if that is the case.

Thanks! Can't wait for it to work again. Good thing i saw this post...
I was starting to think that i'm doing something wrong or that my deck or PC or cable is faulty lol.
Anyway, ill try to be patient. Tanks again! The best plugin hands down, and the most practical and useful.

gittyno1, 2023-12-05:
> Couldn't get it to work even by rolling back to 3.5.5 any tips?

Wait for the update on plugin.

gittyno1, 2023-12-05:
> > > Couldn't get it to work even by rolling back to 3.5.5 any tips?
> > 
> > 
> > Wait for the update on plugin. But for a temporary solution, boot previous update, since new deck update doesnt work. 3.5.5 works. To do this shut down deck completly, than hold 3 dots button and tap power while holding 3 dot button till you see boot menu. release 3 dot button and select "previous" option (for me its secind option and let it do its thing and it will boot into previous update that worked with DeckMTP). I say this is temporary because when you restart the deck it will go back to curent update, the new one. But doing this is a workaround because you can use it to transfer what needs to be transfered till dafta fix the plugin for new updates.
> 
> You can actually set it to permanent after you rollback, go to desktop mode and add this in the terminal "sudo rauc status mark-active booted" after you're done, you can reboot to as many as you can, it won't revert back to the latest fw.

Thanks, but i can update the system as normal if i want, correct? This wont break any update features?
Also i still wanna wait for the plugin update.

Archer201977, 2023-12-05:
> Thanks, but i can update the system as normal if i want, correct? This wont break any update features? Also i still wanna wait for the plugin update cause i do like deck updates and how valve improves the deck. I got LCD version by the way.

"Yes you can, won't break, its fine on my end.

gittyno1, 2023-12-24:
Is there any odds that this will be updated?

Archer201977, 2023-12-24:
> Is there any odds that this will be updated to work with 3.5.7 and up?

I think Dafta will work on this, but right now he is probably a lot of stuffs to do so this will be on hold for a while..

dafta, 2023-12-26:
An update is definitely coming, but it probably won't be ready in a while. Before I can update anything, upstream BIOS has to be patched with DRD support again, which, given Valve time, might take a while.

Archer201977, 2023-12-26:
Thank you very much Dafta!

dmrickey, 2024-01-05:
So there's nothing to be done for this? T_T

Archer201977, 2024-01-05:
> So there's nothing to be done for this not working right now? T_T
> 
> I understand there's not much you can do if it simply can't work with the current version, but it's just frustrating because this is the most useful decky plugin and it seems to break every other update :(

Dafta just responted previously that he it will take a while. So he is going to do it but with an unspecified time frame.

Archer201977, 2024-01-05:
@dmrickey - In the meantime you can just use a wireless transfer, just disable the sleep function in you deck to prevent it from disconnecting with your source.

dafta, 2024-01-06:
> So there's nothing to be done for this not working right now? T_T
> 
> I understand there's not much you can do if it simply can't work with the current version, but it's just frustrating because this is the most useful decky plugin and it seems to break every other update :(

Sorry, but unfortunately, no.

I understand the frustration, believe me when I say that I feel it too. Every other BIOS update breaks DRD completely, so this plugin, my tethering script, and anything else I'm working on stops working until they fix it.

If you're feeling brave, you could update your BIOS to 20, but here be dragons.

ibleem, 2024-01-28:
We're really looking forward to the update :)

ScottCarammell, 2024-02-10:
I'm sorry but is there really no better software for transferring files between PC and Deck? I mean seriously. Valve breaks DeckMTP every two minutes, Warpinator is crap and never works, and FileZilla takes way too long to set up to be worth my time. This is why I don't buy games without cloud saves anymore.

dafta, 2024-02-10:
Syncthing is great for syncing folders, heavily recommended for syncing saves and such. KDE Connect is great for sending files. You can also transfer files using rsync or scp over SSH, this is what I use. For transfering games over LAN, Steam now has the feature to share games between devices, but it might be off by default.

Archer201977, 2024-02-10:
> I'm sorry but is there really no better software for transferring files between PC and Deck? I mean seriously. Valve breaks DeckMTP every two minutes, Warpinator is crap and never works, and [...]

DeckMTP won't break, if you disable the sleep option on your deck, sleep causes the deck to breack communication between pc, judging that you are still on older OS?

Archer201977, 2024-03-10:
So i tested the DeckMTP on the latest Steam OS 3.5.17, it seems to work for about 5 minutes being connected, but then suddenly crashes, disconnecting and reconnecting the c-cable works again for 10mins then crashes, i think this will work properly on the coming update hopefully :)

dafta, 2024-06-18:
This needs to be tested again with the 1.0.3 update.

Honzackcz, 2024-06-18:
From my quick test DeckMTP 1.0.3 works with SteamOS 3.5.19 (BIOS F7A0120) only thru USB 2.0 connection tho.

Archer201977, 2024-06-18:
> From my quick test DeckMTP 1.0.3 works with SteamOS 3.5.19 (BIOS F7A0120) only thru USB 2.0 connection tho.

How did you installed the 1.0.3 version? Currently its not on decky.

Honzackcz, 2024-06-18:
> > From my quick test DeckMTP 1.0.3 works with SteamOS 3.5.19 (BIOS F7A0120) only thru USB 2.0 connection tho.
> 
> How did you installed the 1.0.3 version? Currently its not on decky.

In Decky beta options switch store channel to testing, then you should be able to update DeckMTP.

dafta, 2024-06-18:
After updating DeckMTP, be sure to switch the store channel back to default, so that you don't update other packages by accident later.

Archer201977, 2024-06-18:
Cool! Thanks Honzackcz and specially Dafta.

dafta, 2024-06-18:
This is most likely the same issue as #9 , a problem with Windows and MTP drivers. I'm closing this issue to keep the discussion focused on one issue.

closed 2024-06-18 by dafta