dafta/DeckMTP #19: [Enhancement/feature request] USB redirect

DanHolli123, 2024-01-13:
The controller for the Deck internally connects as a USB device. Would it be possible to forward the connection to the USB similar to USBip?

dafta, 2024-01-13:
It's already possible to use USBIP over a cable if you use my script to tether the Deck's internet connection, [here](https://gist.github.com/dafta/0aadeba3aa8bcbbc8b92a233977571ed).

As for forwarding the controller directly, it's theoretically possible, and Valve is definitely working on it, as there is a program on the Deck designed specifically for this, called `usbhid_passthrough` or something similar. I have tested this program, and while it works, it has some issues. For instance, it works perfectly if you don't turn on Steam on the PC, and you use the default controller configuration. However, as soon as you turn on Steam and it tries taking control of the Steam Deck controller, the controller crashes and stops working.

Valve is definitely working on it, though. The program is receiving updates, as recently as a week ago.

DanHolli123, 2024-01-28:
That is fantastic! Maybe a bit of community work can help speed up that development? 😁