dafta/DeckMTP #20: No Deck at Windows 10 Device Manager

Nx0Ri, 2024-01-21:
Using USB-C to USB-C isn't working, ik there's people having same issue, just tested
Using USB-A to USB-C with many cables isn't working, pc ports are working (other devices charge and file transfer with these ports)
Bios: 116
Steam client: Stable latest
Os: Stable Latest
CEF is enabled, DRD is enabled

Bios version isn't 115 cuz some other decky plugins doesn't work (116 is most stable tbh)
Also DeckMTP worked before on 116 (before I reimaged steamos and installed refind + windows 11)

dafta, 2024-01-21:
Unfortunately, BIOS 116 had issues with DRD, leading to Valve disabling DRD in the newest BIOS until they fix it. This plugin is currently broken with the newest BIOS, and there's not much I can do until Valve reenables DRD.

One thing that could potentially help with the issue, you could try uncommenting line 83 in `~/homebrew/plugins/DeckMTP/bin/umtprd.conf`. The line should say `no_inotify 0x1`. Please let me know if that helps.

sylvathemoth, 2024-02-16:
> Unfortunately, BIOS 116 had issues with DRD, leading to Valve disabling DRD in the newest BIOS until they fix it. This plugin is currently broken with the newest BIOS, and there's not much I can do [...]

we did notice there were a couple pre-release bioses on gitlab, we gave 109 a shot on our oled deck, and DRD is fixed!
im assuming 121 is for the lcd deck, and should also be fixed.
https://gitlab.com/evlaV/jupiter-hw-support/-/tree/master/usr/share/jupiter_bios

NightSprinter, 2024-02-18:
I just flashed 121 onto my 512GB LCD model.  I can confirm that DRD is working great again.  The Deck shows up in Explorer as a portable device (though just the SSD, will scrounge the issue tracker for a report, otherwise make a new one for that).

erictan2001, 2024-03-03:
> Unfortunately, BIOS 116 had issues with DRD, leading to Valve disabling DRD in the newest BIOS until they fix it. This plugin is currently broken with the newest BIOS, and there's not much I can do [...]

It works! Tested on both windows and debian 12 kde.

dafta, 2024-03-05:
Newest update of DeckMTP should fix this by default now. Unfortunately I can't update the plugin on decky until it works on Stable, so at least until the newest BIOS hits Stable.

closed 2024-03-05 by dafta

sylvathemoth, 2024-03-05:
my deck got a firmware update a day or two ago on stable, so oled at least
had a push to stable

dafta, 2024-03-05:
That's great to know! I'll check if it's the correct BIOS and update the plugin tomorrow then.