dafta/DeckMTP #21: MTP Session Crashes

amorrn, 2024-03-08:
After transferring a file (no matter the size) the MTP connection seems to hang/go stale. The transfer is successful and the file is complete, but after the transfer completes the MTP client session doesn't seem to refresh to show the new file and I can't navigate anymore. Turning the MTP off and on again on the deck side seems to resolve the issue and allow me to transfer another file (after reopening the MTP client application on my laptop).

This is on a MacOS 11.7.10 client running openMTP and Android File Transfer applications to test (same issue is encountered on both programs).

dafta, 2024-03-08:
This should be fixed with the newest update when it gets tested and uploaded to the decky store, but in the meantime, you could try uncommenting line 83 in `~/homebrew/plugins/DeckMTP/bin/umtprd.conf`. The line should say `no_inotify 0x1`.

amorrn, 2024-03-20:
Thanks for providing the workaround, that worked for me.

closed 2024-06-18 by dafta