dafta/DeckMTP #22: [Bug] Crash after copying an 8 GiB file

aidinzolghadr, 2024-03-08:
Hi,

First, I appreciate your work, this plugin is super awesome. There is a bug that when I'm copying an 8.20 GiB file to Deck, after the copy finishes (I can't see the progress bar as it is filled but the description of the copy window says 15 seconds left), suddenly Steam Deck drive disappears and I have to enable it from plugin menu again.

I uncommented line 83 in config file mentioned in the other issue but didn't help.

This happens both in Desktop and Gaming (?) mode (Deck's default, Big Picture?).

I could not find a log in it's directory to upload, if it's located elsewhere please let me know.

Also let me know if I can be of any help.

Cheers.

dafta, 2024-03-09:
This *might* be fixed with the newest update that's in testing. It's not yet in decky repos, but once it is, could you retest this and see if it's still an issue?

In the meantime, you can grab the logs with the following command: `journalctl -b > deckmtp-log.txt`

Could you recreate the issue and then run the command?
And then upload the `deckmtp-log.txt` file here.

aidinzolghadr, 2024-03-10:
Sure, should I comment that line again?

dafta, 2024-03-10:
No, you can leave it uncommented, it's uncommented in the new update anyway.

closed 2024-06-18 by dafta