dafta/DeckMTP #24: Add USB tether support

DaniGuardiola, 2024-04-28:
On Reddit, you mentioned that you were adding support for USB tethering (your script at tinyurl.com/sdeth) to this plugin. I would absolutely love that (use case is connecting to my meta quest 3 via cable to have a large virtual screen on long flights, etc). Thank you!

dafta, 2024-04-28:
USB tether support, and other USB functions, are definitely on the roadmap, and I will be adding support as soon as possible. Unfortunately, it's hard working on anything USB related because Valve breaks USB DRD every other update, causing this plugin to stop working. USB DRD has been broken for some months now, and I'm still waiting for Valve to fix it before any new versions can come out.

dolphin-evorobey, 2024-06-27:
@dafta hola mate, any good news? very lack this plugin

dafta, 2024-06-27:
This is the very next thing on my list. The `systemd` branch of the plugin is mostly complete, which was the requirement for most of the modularity I intend to offer in the plugin, and soon adding new USB gadgets will be a lot easier. Tether support is the first one of these I intend to implement.