dafta/DeckMTP #25: DeckMTP not working

hydroxi1, 2024-05-05:
Tried the stable version, and both versions in the test store, neither work even though a month ago the 1.02 one worked, currently on SteamOS 3.5.19
Is the plugin not updated or is it something on my end?

dafta, 2024-05-05:
It's not you, it's Valve. This project in in a kind of limbo right now, where every new SteamOS update has the potential to break USB DRD which is a requirement for the plugin to work. Unfortunately, there's not much I can do about it, DRD is currently broken until Valve get to fixing it.

TalosDx, 2024-05-25:
I found out that there is a slower alternative, it is sftp, but in all the guides they write "enable" sshd. And it's nonsense, because a normal user will be walking around say in a cafe and using unsecured wifi with ssh sticking out.... sounds like the authors of such guides decided to create a botnet, lol.

In general I propose to make a plugin like deck mtp, only deck sftp. In which add a simple checkbox to control whether sshd is running or not. In addition, add a 30 minute timer to automatically disable ssh. 

That way users would at least realize that keeping ssh enabled is not very secure.

I would codify an open sftp client on kotlin, which would force the user to use rsa keys. 

I measured the data transfer speed and for some reason mtp is not much faster than samba, sftp has not been tested yet.

It's a pity, of course, that mtp can't be fixed yet, still it's more convenient than samba or sftp.

In any case, I use the opportunity to say thank you to the author, you're really cool man.

dafta, 2024-06-18:
The build on the testing store should work now, and will hopefully reach stable soon.

closed 2024-06-18 by dafta