dafta/DeckMTP #26: Cannot read propertiesof undefined (reading Title)

lucasmerlin, 2024-06-16:
I'm getting this error when when tryning to use DeckMTP. Maybe caused by a recent SteamDeck update?
I just installed decky loader, so no other plugins are installed. I did enable the usb drd mode.

lucasmerlin, 2024-06-16:
I guess it's related to this: https://github.com/SteamDeckHomebrew/decky-loader/issues/597

dafta, 2024-06-17:
The issue is already fixed on testing, but the fix won't reach stable for a while since the plugin just doesn't work in general because USB DRD is broken. Unfortunately, I cannot tell you when this will be fixed, as it depends on Valve fixing USB DRD.

dafta, 2024-06-18:
The plugin on the testing store should work now, if you want you can try to install from testing. Otherwise, this should hopefully reach stable soon enough.

closed 2024-06-18 by dafta