dafta/DeckMTP #29: Device detected but empty directory

Br3nnabee, 2024-07-03:
Running DeckMTP for the first time, latest beta steamOS and latest testing version. DRD is enabled, steam deck is detected as a device by my windows 11 PC, but Home directory is empty and I'm unable to transfer files.

ChikaraAra, 2024-07-03:
Just to ask, I just got my Steam Deck and Deck MTP doesn't work and give me a TypeError something, If u had it, How did u fix it, ty in advance.

Br3nnabee, 2024-07-04:
> Just to ask, I just got my Steam Deck and Deck MTP doesn't work and give me a TypeError something, If u had it, How did u fix it, ty in advance.

This is not related to the issue here. You need to switch over to the testing version and there's issues about this already made.

dafta, 2024-07-05:
That's curious. Could you try different USB ports, and a USB 2 cable? To rule out that it's the same driver issue as the other bug.

If it still happens, could you send me the output of `journalctl -b`?

dafta, 2025-01-22:
Closing this as duplicate of #31

closed 2025-01-22 by dafta