dafta/DeckMTP #30: TypeError on install/boot/etc

codefaux, 2024-08-03:
Reproduce;
- Factory Reset Steam Deck
- Install Decky
- Install DeckMTP
```
TypeError: Cannot read properties of undefined (reading 'Title')
  at eval (eval at importReactPlugin (http://localhost:1337/frontend/chunk-<etc>
```

PowerTools and other plugins work

Casuallynoted, 2024-08-04:
Same issue on my deck, running the latest SteamOS/Steam

dafta, 2024-08-04:
You need to install DeckMTP from the testing store, because the update still hasn't reached Stable. Here are the instructions:

Go into decky settings, and under store channel, choose "Testing". You can then go to plugins and update only DeckMTP. After that, you can switch back to the Default testing store. Decky will notify you that you can update deckmtp but it's a downgrade to the store version, so just ignore it for now.

If you find that the version in the testing store works, and you'd like to see that version in the Stable decky plugin store, I'd kindly ask you to leave a comment on the PR [here](https://github.com/SteamDeckHomebrew/decky-plugin-database/pull/604#event-13717719061) using the testing report template from [here](https://wiki.deckbrew.xyz/user-guide/testing#submitting-a-test-report).

closed 2024-08-04 by dafta

codefaux, 2024-08-05:
Fantastic, glad to see it. Report sent, all good!

Closing thought: might be worth putting the note about BIOS > DRD in the DeckMTP store description and/or plugin UI as I found the plugin from the store rather than from github.

Thanks for your work!

dafta, 2024-08-05:
> Closing thought: might be worth putting the note about BIOS > DRD in the DeckMTP store description and/or plugin UI as I found the plugin from the store rather than from github.

This is already in the plugin UI. If the plugin detects that DRD is off, it shows the instructions to turn it on. You can see the message it prints [here](https://github.com/dafta/DeckMTP/blob/3ff8b6522417849c2e21d11e59ce9968445e617b/src/index.tsx#L45).