dafta/DeckMTP #33: Able to connect but no files visible

ven-toy, 2025-03-14:
I want to connect my PC with my Deck through USB-C which I'm able to but I can't see any files and I can't copy from my PC to the deck either.

dafta, 2025-03-14:
Hi, can you try the workaround mentioned in the comment [here?](https://github.com/dafta/DeckMTP/issues/31#issuecomment-2599861596) It should hopefully fix the issue.

If it does, please comment back so that I know to push that update as soon as possible.

ven-toy, 2025-03-14:
Worked!

closed 2025-03-14 by ven-toy

dafta, 2025-03-27:
Hey, so the new version that fixes this issue is on the testing store right now. Before that gets merged, it needs testers. If you have the time, I'd appreciate if you could install the version from the testing store and leave a testing report on the pull request for the new version. Here are the instructions:

Go into decky settings, and under store channel, choose "Testing". You can then go to plugins and update only DeckMTP. After that, you can switch back to the Default testing store. Decky will notify you that you can update deckmtp but it's a downgrade to the store version, so just ignore it for now.

If you find that the version in the testing store works, and you'd like to see that version in the Stable decky plugin store, I'd kindly ask you to leave a comment on the PR [here](https://github.com/SteamDeckHomebrew/decky-plugin-database/pull/786) using the testing report template from [here](https://wiki.deckbrew.xyz/user-guide/testing#submitting-a-test-report). I'd appreciate the help getting this plugin into stable finally.